R0rt1z2/realme-ota #5: help me

qiurigao, 2021-12-30:
realme-ota -c 2 -d 0 RMX2202 RMX2202_11.A.09_0090_202106051941 2
[2021-12-30 23:42:27.392070] I: RealmeUI V2 RMX2202 (A) - 09_0090_202106051941
[2021-12-30 23:42:27.394198] F: This shouldn't happen. Something went wrong while requesting to the endpoint (Failed to parse: https://component-ota-in.coloros.com/update/v1)!

R0rt1z2, 2021-12-30:
> 

The command works fine for me, not sure what are you doing wrong:
```json
[2021-12-30 16:53:13.801115] I: RealmeUI V2 RMX2202 (A) - 09_0090_202106051941
{
    "aid": "RMX2202NV1B_11.A",
    "androidVersion": "Android 11",
    "colorOSVersion": "ColorOS 11.2",
    "componentAssembleType": true,
    "components": [
        {
            "componentId": "20664_system_vendor_20664.1.34.202112071136043849361",
            "componentName": "system_vendor",
            "componentPackets": {
                "id": "foreign_20664_system_vendor_20664.1.34.202112071136043849361_1_97d99bed062c31c989a10ccbcf19160e",
                "manualUrl": "https://gauss-componentotacostmanual-in.allawnofs.com/remove-c649c0c4d7d1124c8e6a08afb5476eb7/component-ota/21/12/08/eeb2a7806dcd4ab6a4dbac82d21fa237.zip",
                "md5": "97d99bed062c31c989a10ccbcf19160e",
                "size": "4097040772",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-in.allawnofs.com/remove-c649c0c4d7d1124c8e6a08afb5476eb7/component-ota/21/12/08/eeb2a7806dcd4ab6a4dbac82d21fa237.zip",
                "vabInfo": {
                    "data": {
                        "extra_params": "metadata_hash:3f5f681991652568fae64ec129dc34fda76e8bf8645f7670bb9a9f5c7218c42e",
                        "header": [
                            "FILE_HASH=W0yivqfpeVcsFi6gbp4KBG4Hc8ZB8bF7q9o3YdgZZVg=",
                            "FILE_SIZE=4097036873",
                            "METADATA_HASH=4chH478JeFIp44+UuHALX67w9ydiJkyfmHwNwJuPxe4=",
                            "METADATA_SIZE=200596",
                            "ota_target_version=RMX2202_11.A.17_0170_202112071503"
                        ],
                        "otaStreamingProperty": "payload_metadata.bin:704:200863,payload.bin:704:4097036873,payload_properties.txt:4097037635:209,metadata:69:588",
                        "vab_package_hash": "97d99bed062c31c989a10ccbcf19160e"
                    }
                }
            },
            "componentVersion": "20664.1.34.202112071136043849361"
        }
    ],
    "decentralize": {
        "offset": 5894,
        "round": 28800,
        "strategyVersion": "default"
    },
    "description": {
        "share": ".",
        "url": "https://gauss-componentotacostmanual-in.allawnofs.com/remove-c649c0c4d7d1124c8e6a08afb5476eb7/component-ota/21/12/23/eeb66ac69a654e7680fd3098c2ae0e7a.html?logoType=1"
    },
    "googlePatchInfo": "0",
    "group": "RMX2202_11.A.17_0170_202112071503",
    "id": "61cbaf1f66c96100cfa17a7e",
    "isNvDescription": false,
    "isRecruit": false,
    "noticeType": 0,
    "nvId16": "NV1B",
    "osVersion": "ColorOS 11.2",
    "otaTemplateId": "61c59f4b66c96100cfa15522",
    "otaVersion": "RMX2202_11.A.17_0170_202112071503",
    "paramFlag": 1,
    "parent": "ota-template",
    "reminderType": 0,
    "rid": "e9f4f8c1-8052-4d66-9d4c-cfb84ec4fc0b",
    "securityPatch": "2021-12-05",
    "securityPatchVendor": "2021-12-05",
    "silenceUpdate": 0,
    "status": "published",
    "timestampH5": "2021.12.23",
    "versionCode": 170,
    "versionName": "RMX2202_11_A.17",
    "versionTypeH5": "Official version"
}```

cloverkits, 2022-01-02:
Have the same problem 😂
```
➜  Desktop realme-ota -c 1 -t 5 -d ota.txt RMX3161 RMX3161_11.A.19_0190_202112201219 2
[2022-01-02 23:01:04.494026] I: RealmeUI V2 RMX3161 (A) - 19_0190_202112201219
[2022-01-02 23:01:04.560684] F: This shouldn't happen. Something went wrong while requesting to the endpoint (HTTPSConnectionPool(host='component-ota-cn.coloros.com', port=443): Max retries exceeded with url: /update/v1 (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1129)'))))!
```

product_model: RMX3161
ota_version: RMX3161_11.A.19_0190_202112201219
Realme UI Version: V2.0

R0rt1z2, 2022-01-02:
> Have the same problem joy
> 
> ```
> ➜  Desktop realme-ota -c 1 -t 5 -d ota.txt RMX3161 RMX3161_11.A.19_0190_202112201219 2
> [2022-01-02 23:01:04.494026] I: RealmeUI V2 RMX3161 (A) - 19_0190_202112201219
> [2022-01-02 23:01:04.560684] F: This shouldn't happen. Something went wrong while requesting to the endpoint (HTTPSConnectionPool(host='component-ota-cn.coloros.com', port=443): Max retries exceeded with url: /update/v1 (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1129)'))))!
> ```
> 
> product_model: RMX3161 ota_version: RMX3161_11.A.19_0190_202112201219 Realme UI Version: V2.0

Interesting, even though that's a completely different issue, looks like they changed the CN server.
Assuming you have a Realme with CN region, could you dump `/my_region/etc/extension/appfeature.region_in.android_coreservices.xml`?

cloverkits, 2022-01-02:
> Interesting, even though that's a completely different issue, looks like they changed the CN server. Assuming you have a Realme with CN region, could you dump [...]

No file found `/my_region/etc/extension/appfeature.region_in.android_coreservices.xml`
Found `/my_region/etc/extension/appfeature.region_cn.android_coreservices.xml`
**File Content:**
```

cloverkits, 2022-01-03:
> Interesting, even though that's a completely different issue, looks like they changed the CN server. Assuming you have a Realme with CN region, could you dump [...]

```
POST /update/v1 HTTP/1.1
language: zh-CN
romVersion: RMX3161_11_A.19
androidVersion: Android11
colorOSVersion: ColorOS11.1
infVersion: 1
operator: 
otaVersion: RMX3161_11.A.19_0190_202112201219
trackRegion: CN
uRegion: CN
model: RMX3161
mode: 0
nvCarrier: 10010111
partCarrier: 10010111
localCarrier: 10010111
Content-Type: application/json; charset=utf-8
Content-Length: 975
Host: component-ota.coloros.com
Connection: Keep-Alive
Accept-Encoding: gzip
User-Agent: okhttp/3.14.4

{"params":"YnVteDHdGV42PIGQLW2xYKZZLuI9\/r1aLfS5laubkKR1bIBY7ale8dT6ruGgNbrNWfSqGbzWIHcdT8va\/R6usQJ0m7FcII2xmCfYW4xxl0El8Z326nbnTeSiNaO91ad3U41IswaT55DqhaSEKsBu9O+kPmMNlrp\/d3DSEjdXYfw8lLDfENbm+D6UGWg5dcSJd6gM15fURIVJ7E9TrlThixC2L4fwA44B8+XzMsGVdglODsbWy9eooGIelWTPqgQ+sIuFAdhJdyvUR69F\/C5Jds\/\/cWGb0OCPosTKApz68uWx9\/yxDPU0tdyJz5UlegtAquJpBGiyAm6zCTe8cetF\/1l5nGzxeIUvjNjewYQZ8DdpwbaXdcSGzblpR0RMiH9Deyw7pI9bT1IZwRq8+7xUtXEAXTQUQXJl\/FBNq7Do1xM\/e06j\/U+WkLJkGiNsc8fH2QDmzxg6UjEKXSlns62s6w5BwsxBl2M6px9cj7m1Y\/QnjQ2pTxERPcbxYNFHwHTW1L4tTlDQzVR1h2xVTtbdeIajYhv6nqodwm3eCMa6V9+TEQrkSf49jcKPoWGqqXqqXHlCCZBLZKj0pGeUdSbhBCj1QLoZqMn14ylttW59hcrMOwklBSwq5TARI77U0mC7tJT4k7z\/rFVFajlnaJwYmz\/CxAVre83\/r4mYeLA\/eOROjjbQHUQSPzrkQk6Fc57g150o\/HC2+7SsdjBMdRQvyc9rYKIruc0FeKpFBt046bKPxziI9WEx+HoAleRlP8MpA1\/dNkidtSdyZCvVb+L7IP7WkbzPj7F42vZ7xZOU4gm\/t9zgkcXQwXdy3RZPVN8vUSIIDPPOUFuWsjEnnuHnH3hMhj97vZbNX98zdMew8gd4QrLoImuOjotBVcU5EEM4XKVQza4vSR+axIiQ7AOcEXto9iVRWQ==2m0sYCA%fbTuAzi"}
```

R0rt1z2, 2022-01-06:
> > > Have the same problem joy
> > > ```
> > > ➜  Desktop realme-ota -c 1 -t 5 -d ota.txt RMX3161 RMX3161_11.A.19_0190_202112201219 2
> > > [2022-01-02 23:01:04.494026] I: RealmeUI V2 RMX3161 (A) - 19_0190_202112201219
> > > [2022-01-02 23:01:04.560684] F: This shouldn't happen. Something went wrong while requesting to the endpoint (HTTPSConnectionPool(host='component-ota-cn.coloros.com', port=443): Max retries exceeded with url: /update/v1 (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1129)'))))!
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > product_model: RMX3161 ota_version: RMX3161_11.A.19_0190_202112201219 Realme UI Version: V2.0
> > 
> > 
> > Interesting, even though that's a completely different issue, looks like they changed the CN server. Assuming you have a Realme with CN region, could you dump `/my_region/etc/extension/appfeature.region_in.android_coreservices.xml`?
> 
> No file found `/my_region/etc/extension/appfeature.region_in.android_coreservices.xml` Found `/my_region/etc/extension/appfeature.region_cn.android_coreservices.xml` **File Content:**
> 
> ```
> <?xml version='1.0' encoding='UTF-8' standalone='yes' ?>
> <extend_features>
>     
>     <app_feature name="com.coloros.securitypermission.enable_personal_info_protection" args="boolean:true"/>
>     
>     <app_feature name="com.coloros.securitypermission.intercept_lockscreen_showsettingui" args="boolean:true"/>
> </extend_features>
> ```

We figured out the problem with the URL already. However, now we need the CN props to properly craft the request. Would you mind sending me the output of `adb shell getprop` (feel free to censor whatever you think it's sensitive).

qiurigao, 2022-01-06:
> 
[rmx2202.txt](https://github.com/R0rt1z2/realme-ota/files/7821995/rmx2202.txt)
cn

qiurigao, 2022-01-06:
HTTP/1.1 200 OK
Server: nginx
Date: Thu, 06 Jan 2022 05:30:08 GMT
Content-Type: application/json;charset=UTF-8
Content-Length: 3263
Connection: keep-alive
Access-Control-Allow-Origin: *
X-Backend-Host: 0025:8099
X-Gateway-Host: f1cc5e4ba262a4c1a67c18c8f4fc15813f49ef0cb03627023ff6a5beac0b0590aaa6a69672bee83974fc020caf87b5521ea00b3cccfa0b3cd931f6745a779c16

{"body":"aWocPwwx21jET0pfpeynWV3xhiQwZSx/3pH7das2GvF3mTFoie4lnZ8h+Cmt4c3KbEqZNpFYNita8FkXpF2Z0AbgELEvsxsrPBCRUajyEtO9PB+y1HWfn/m/NwveRkiAbAn+p9P5LhRMUIXxybi4Dd+KN1+xCrfICt6GyW6r5DJKJGAtBu8EBJb7dfxEg5jNQXay//5unfg5/l0AMS6UkX+G1nzL6C5PC8ui5z2R02zUiDrQZ38n8D6gJYDiCGFLOua6l8/z13kDai5U90AJrykUna6bwo0E9nv8yrw86iKO6jsqIZopD5ALq4WNDrW6z37zwT55xk9bD9fg3Yyu+MNkONijjfodVQ8uWv7oeH+NGEap8qBAZG4gNqbikACpdYAAhtvOaoqiBUugoM8Jn7kbD0RnynYupcthCp46AJh9LmSo6czQF8J+BIMZpoT2gp6WTbrOiaWUJIqJy4ejHYDnS0maQfig27jUerT9N3gXrK5jkCdt5zxC+lNbwPgeRMThWJ2uuEWf8it0Ky0Gb2b8nMBPZWbkGJLlD3gcbybse7omHffEZ1FNZ+DHMQah7C3Ilk32/n/Gui4g3dI4omcgS9wEyE/8TDIFGAgo/tz3k1EbO/P687m6k3F6GamQT/fuh74wZpYi7Q2wUXw7JBpVxHU6chNs1aie9BMj0d0sTUZ+4K89uueDgnZl5YlOPzn6Yq2DrbZEsp4/ZlBQ2s43TmU5FHhGNw83qvfgm8Pym9SrDgQdElWA3VDJ6S0VeMoPEYMemtCK73vIS4VzNRvsRKFI3ocmJj/WAOnkoH/71LeGHZsKr552PicDZoib9dzUH/GcqyAldld8quJq/Qso/vkKTenEd+BBbrjZM+T8beJRBw5jXUyOKPvHMyl8sfqMKgNNCEYg415C3xl2teMrCvgKYk6jZKtOVVoVT/Vdg55f6nJsCv/JX3Z9CmnGOWvgRSGhPE7SRE3RwLIjsKYejRa1JRE7ZKEn1ED98V5BEmU5ALirbvWqDLH/SCfzb8WOxjV0fnXk+QV7heGFtuTnQXD//Sqbrb3z8OpMWaqgsci/l6h/u6N8jXf13hvNvN+TlQ8nMyKYRq2r5knJEyi+H9/7dcv7e5DF1NsArX5IDYIYj5tFU3uqxzLLR0kjhn2fPnLZ7Z6eXoSCp5tarctWf72NVclHeE7yRQJ6DBupAkfy/2ultIyP3JueeWhcdDvaG3wjO5VJmnNrViPy7m4MhuRQ8FhAHXa64gq7D+Qnokx0+G2lLI06Cafm6YoSKUe3Eut9aloqcJf5d/wQG7eTiezRbhgi0slWGub6sGWnmzBhu4SSFI672b1ozwILittmxfef0CyQGDkG+4lXewNSa10rNWGicmt5CMmCkFESdFCOrs5cYkEFYMkAa35ZdDakuwKow2uXRPe0AyPjT1gIUE1hO0iat3S1FlsyU1UoexPQorfSkOQV/vI9CQRkcdIa6TW6jEBi/RNOz1VZ9eR+qa5adGOXSB2/rq1A5wkZFQ6lYlYmxC0PA1YQgAVrSJpUYotSCy8wmEaR9sHuDcgy6qSRge/VOeJEhLzLnVr+TtYpxM4up08m4mjIoYLYv2NEfR9SzKOtrMyBbJNXjeWopyaJbCTFYmoGTtwsIYwvaUd8jdAOKyuT87or7ZlsX9ILJsP7u8YMvM+hNcSRMRuao+7AUFGKtDkxSrJj0h0b7qwgv6dFXBDD1OuIzb8hW6xTa0SMew2eVRxDlA6g90K3HA48zXs4GR3FiyedUWzlSH6E5lMYfJDSbx6vSyC1rI57aI2SQqDe6vLBg5D21/UzKPE5W4CAZsI3KxLj1gfRdxYf3881wYddB8Vy+NM+xHarJOkGTWIEPmNRHDUnRVh6QaxMf0G0SUPZo30zvKR+QWJ7TQWBthqEGPRwX62zhQfIJP21ZE17DeXb9lQq9IKvhI4Z94ou9HmgwAQKC6Lf3ur3JW/zYOiFYRmiu3XBgR37o2orSRl0dJOg7pj+fpNkMJmTKFD2hwbaij1ADiNnXIp8aXHxctSx/wi2pveUI+4MGTUMvVkBB7W1JBNlzfFGJZBs5PaFyOBoFZKbNeYb1AeSK0Al+VAsDfL8AbQ4l9DkDisAGVee5dKZXlP9hT52pSYs2ScwSj6LhlVkc3IhwCZm0BEQ+xPCNDHkdrwgAIDrk361yKoIa3TaddbAj3fj5LxnlvgDLnrrl/ztzD3nhriR+58DGKUIRh1EL0VOA4DHSUXZWtzXXmvitkVz6wFkocz9LsXVeIIHOdH0xO2XqMIOJG4YseXCOFr7shWW00PIz66DXykhoc+Jp5jy3wcJuIsk2H4NlIxuUfO8FpagtvOwFBlDH05sZaOqhBRHXhy4CMIvt2z5yC1wQ2f7RLS1gZBd2itwBoufezwM1bX68p6i0OWsLNGH8HfqJjHPAL/FEnyb9zafE9hKRTUbAdknjaMFusprJmJLqc6PeDcdc/cmnYAEBJcOl72FZ3S9tiYA1JrdqMWz4wdZqlDBF6ojjZDWMCYBqjxBOBHwVMvtPjiHh6s4MeqfCfoSG/x1ndV03viRDHJc5ZkQPG0X0011M01u7zk/K/QfXnuR2f+yx3qSY/Zi841IkvsKucVYt33t+lvgLgB+Q3/DmvJF0e5Sl7T06+pbSveKmR/XAunZdnAKD/GQ7rFFBAmltQDIjcujhe3AHzN7cX9idjOv+B6UKAbRSItdXj53/cFhoMhMEFyOmT5c7s+2s+JHsgA3Jyjvh5zOhgkqrPno6qpfEsan77vQXCRrYxvozIC/kKWI8FvJeO+jI2ElrtkAkMYuzNAbTWdaKR0Oz3B1FeRuKRSv1dfTNLbO3KeqkVl43Mi6+I95UtHLjy4p0ld22GSohbg6xMcpdDKLuZIYWOgnbTnE40/qvSg+hqyIrt+QvVcWCjBnHvFcwgMdlueF6QXlHG+2ap0Hqir3Cc2r0mj+Lb/CMInpa8Yzfzt25HE/AwPNdMIzFizaJLKvk+1k03PnZZvfSZHeX79E/tlEXQJ/rMCnf7OcbEHKfAzqjjYi8ph/XBj/hFKPYAU2/TF2mKeCn9hQYXTiS4ojACUMhnruZgIEIi2rSzsoR13ayWhFWSWeR9ujj4KPbawuVNiDYnrKt/KcF0upiEMS7HoMIcOraEOJIOYIXu8NRHLXWsDmABgCjCb6eiJfRiypYmP8FraoEp5e7NbecYih8Y_fRA6","errMsg":null,"responseCode":200}

R0rt1z2, 2022-01-06:
> aWocPwwx21jET0pfpeynWV3xhiQwZSx/3pH7das2GvF3mTFoie4lnZ8h+Cmt4c3KbEqZNpFYNita8FkXpF2Z0AbgELEvsxsrPBCRUajyEtO9PB+y1HWfn/m/NwveRkiAbAn+p9P5LhRMUIXxybi4Dd+KN1+xCrfICt6GyW6r5DJKJGAtBu8EBJb7dfxEg5jNQXay//5u [...]

Interesting, this looks like a successful response form the CN server. How did you get that response?
```json
{
   "parent":"ota-template",
   "components":[
      {
         "componentId":"my_manifest_RMX2202_11.C.10_1100_202112161843.97.98cb8dc7",
         "componentName":"my_manifest",
         "componentVersion":"RMX2202_11.C.10_1100_202112161843.97.98cb8dc7",
         "componentPackets":{
            "size":"4791373562",
            "vabInfo":{
               "data":{
                  "extra_params":"metadata_hash:f942a08f7b63e8924014ad1248387351fec421e5627dc414e9b232e28f5b1dbe",
                  "otaStreamingProperty":"payload_metadata.bin:2311:187537,payload.bin:2311:4791367852,payload_properties.txt:4791370221:209,metadata:69:865,metadata.pb:1002:1242",
                  "header":[
                     "FILE_HASH=TYpwstQNUXPHqHHmNfj8I622mtOl2Fn1bFkpriAhi78=",
                     "FILE_SIZE=4791367852",
                     "METADATA_HASH=/c0pOLTPE9LRwWmPa23GTFbBk3/l67s3IR83EAQDlIA=",
                     "METADATA_SIZE=187270",
                     "ota_target_version=RMX2202_11.C.10_1100_202112161843"
                  ],
                  "vab_package_hash":"b0d722c4649d4d04738ce9d5d609213e"
               }
            },
            "manualUrl":"https://gauss-componentotacostmanual-cn.allawnfs.com/remove-6d434f999ae265a31f5ee26bf4dfba14/component-ota/21/12/17/a389e4d165444cffaab4939f63d7c6f1.zip",
            "id":"domestic_my_manifest_RMX2202_11.C.10_1100_202112161843.97.98cb8dc7_1_b0d722c4649d4d04738ce9d5d609213e",
            "type":"1",
            "url":"https://gauss-compotacostauto-cn.allawnfs.com/remove-6d434f999ae265a31f5ee26bf4dfba14/component-ota/21/12/17/a389e4d165444cffaab4939f63d7c6f1.zip",
            "md5":"b0d722c4649d4d04738ce9d5d609213e"
         }
      }
   ],
   "securityPatch":"2021-12-05",
   "otaVersion":"RMX2202_11.A.10_1100_202112161843",
   "isNvDescription":false,
   "description":{
      "share":".",
      "url":"https://gauss-compotacostauto-cn.allawnfs.com/remove-6d434f999ae265a31f5ee26bf4dfba14/component-ota/21/12/22/d7ff0d8db30043f1af4570f1f7f36cf5.html?logoType=1&fontType=opposans"
   },
   "versionName":"RMX2202_11_C.10",
   "rid":"e1e344ac-dd79-4a97-8f7d-0cf0aee9a48b",
   "isRecruit":false,
   "osVersion":"ColorOS 12.0",
   "componentAssembleType":true,
   "googlePatchInfo":"0",
   "id":"61d64dc5124eec00cf8316b4",
   "group":"RMX2202_11.A.10_1100_202112161843",
   "colorOSVersion":"ColorOS 12.0",
   "timestampH5":"2021.12.22",
   "nightUpdateContent":"系统版本升级，或许包含安卓补丁、新功能以及市场反馈问题的修复，建议进行升级以提高系统稳定性，提高您在使用过程中的体验。",
   "paramFlag":0,
   "reminderType":0,
   "noticeType":0,
   "versionCode":1100,
   "silenceUpdate":0,
   "otaTemplateId":"61ce7da566c96100cfa18efa",
   "securityPatchVendor":"2021-12-05",
   "androidVersion":"Android 12",
   "versionTypeH5":"正式版",
   "aid":"RMX2202NV97_11.A",
   "nvId16":"NV97",
   "status":"published"
}
```

R0rt1z2, 2022-01-06:
@qiurigao @cloverkits Try again with the latest commits please. I could dump `RMX3161` (CN) and `RMX2202` (CN) without problems. Please note that you'll probably get rate limited (_Flow Limit_ error). If you get rate limited, wait a few minutes before requesting again:
-> `realme-ota RMX2202 RMX2202_11.A.09_0090_202106051941 2 -c 1`
-> `realme-ota RMX3161 RMX3161_11.A.01_0010_202104240058 2 -c 1`

cloverkits, 2022-01-06:
> @qiurigao @cloverkits Try again with the latest commits please. I could dump `RMX3161` (CN) and `RMX2202` (CN) without problems. Please note that you'll probably get rate limited (_Flow Limit_ [...]
All right. Thank you very much.

qiurigao, 2022-01-06:
ok, good
rui2 pass
but rui3 error
root@105:~# realme-ota -c 1  RMX2202 RMX2202_11.C.10_1100_202112161843 3  [2022-01-07 01:40:24.354109] I: RealmeUI V3 RMX2202 (C) - 10_1100_202112161843
[2022-01-07 01:40:25.433609] E: Request was invalid: artifactV1Result is empty (invalid product name/version)!
root@105:~#

R0rt1z2, 2022-01-06:
> ok, good rui2 pass but rui3 error root@105:~# realme-ota -c 1 RMX2202 RMX2202_11.C.10_1100_202112161843 3 [2022-01-07 01:40:24.354109] I: RealmeUI V3 RMX2202 (C) - 10_1100_202112161843 [2022-01-07 01:40:25.433609] E: Request was invalid: artifactV1Result is empty (invalid product name/version)!

Has your device officially received the RealmeUI 3 (Android 12.0) update?

qiurigao, 2022-01-07:
I have links to some versions.  I can install them normally, I think I need to capture the package to provide you with data
RMX2202_11_C.10

my_manifest
https://gauss-compotacostauto-cn.allawnfs.com/remove-7011767bfd9d533c87386d19ace9e0d1/component-ota/21/12/17/df3ee3bd87a748c593c0760e460854f5.zip
https://gauss-compotacostauto-cn.allawnfs.com/remove-7011767bfd9d533c87386d19ace9e0d1/component-ota/21/12/17/df3ee3bd87a748c593c0760e460854f5.zip
Description Panel URL
https://gauss-compotacostauto-cn.allawnfs.com/remove-7011767bfd9d533c87386d19ace9e0d1/component-ota/21/12/22/c69631be5b884f1285a3407f32160c45.html

Description URL
https://h5fs.coloros.com/a161a6970794463ba6b57ff8d2fe3fab/static/index.html#/about

techyminati, 2022-01-07:
@qiurigao how did you pulled these links?

yuxuxin233, 2022-01-08:
Unable to pull the latest v3.0

R0rt1z2, 2022-01-08:
> Unable to pull the latest v3.0

What model are you trying to fetch for? Indian variant, China variant?

yuxuxin233, 2022-01-08:
CN2022

>

yuxuxin233, 2022-01-08:
CN2202😂