R0rt1z2/realme-ota #8: Received invalid response (artifactV1Result is empty)!

yuxuxin233, 2022-01-14:
[image]

yuxuxin233, 2022-01-16:
[image]
[image]
[image]
[image]

yuxuxin233, 2022-01-16:
[image]
[image]

yuxuxin233, 2022-01-16:
/my_manifest/divisional ro.build.version.ota=RMX2202_11.C.10_1100_202112161843Unable to get

R0rt1z2, 2022-01-18:
Duplicate of https://github.com/R0rt1z2/realme-ota/issues/5 I guess. @yuxuxin233 Has your device officially received the RealmeUI 3 (Android 12.0) update?

yuxuxin233, 2022-01-18:
yes

R0rt1z2, 2022-02-18:
Closing as per https://github.com/R0rt1z2/realme-ota/issues/5#issuecomment-1044675347

closed 2022-02-18 by R0rt1z2