R0rt1z2/realme-ota #9: [EU] OTA is not downloaded properly for the Realme GT Neo2  (Dump file provided)

mmetharam, 2022-01-18:
Terminal just list a bunch of file components:
{
    "aid": "RMX3370NV1B_11.A",
    "androidVersion": "Android 11",
    "colorOSVersion": "ColorOS 11.2",
    "componentAssembleType": true,
    "components": [
        {
            "componentId": "my_manifest_00011011.202201112316474729878",
            "componentName": "my_manifest",
            "componentPackets": {
                "id": "foreign_my_manifest_00011011.202201112316474729878_1_8f235f4e99ee47efa7edf3d49ab0fd85",
                "manualUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/162c6d8c8378419f9369f605012c4754.zip",
                "md5": "8f235f4e99ee47efa7edf3d49ab0fd85",
                "size": "1208283",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/162c6d8c8378419f9369f605012c4754.zip"
            },
            "componentVersion": "00011011.202201112316474729878"
        },
        {
            "componentId": "my_stock_2169.9.26.202201112316474729878",
            "componentName": "my_stock",
            "componentPackets": {
                "id": "foreign_my_stock_2169.9.26.202201112316474729878_1_a17226278b571604f40d0a61994595d3",
                "manualUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/d923bd5ea8c34ce9b2540b222f5fc12f.zip",
                "md5": "a17226278b571604f40d0a61994595d3",
                "size": "742444257",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/d923bd5ea8c34ce9b2540b222f5fc12f.zip"
            },
            "componentVersion": "2169.9.26.202201112316474729878"
        },
        {
            "componentId": "my_heytap_11.2.1.202109092157382762310",
            "componentName": "my_heytap",
            "componentPackets": {
                "id": "foreign_my_heytap_11.2.1.202109092157382762310_1_d514ad3e1838e59d4fa3c210a5b6929f",
                "manualUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/b9f8c005e4504a4986f0f188fc616078.zip",
                "md5": "d514ad3e1838e59d4fa3c210a5b6929f",
                "size": "500084134",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/b9f8c005e4504a4986f0f188fc616078.zip"
            },
            "componentVersion": "11.2.1.202109092157382762310"
        },
        {
            "componentId": "my_carrier_20211021102052.1036395840ab0dca",
            "componentName": "my_carrier",
            "componentPackets": {
                "id": "foreign_my_carrier_20211021102052.1036395840ab0dca_1_e4bdb0cc57bd277606fd1e23deaa61d1",
                "manualUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/7c065d8231684c90aa2e131a83dc90a2.zip",
                "md5": "e4bdb0cc57bd277606fd1e23deaa61d1",
                "size": "1093068",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/7c065d8231684c90aa2e131a83dc90a2.zip"
            },
            "componentVersion": "20211021102052.1036395840ab0dca"
        },
        {
            "componentId": "2169A_system_vendor_2169.1.11.202201112316474729881",
            "componentName": "system_vendor",
            "componentPackets": {
                "id": "foreign_2169A_system_vendor_2169.1.11.202201112316474729881_1_55857667604ffd4a1c632518208d81c8",
                "manualUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/19fd8ae09ed64af1bc13a2e3aef32bec.zip",
                "md5": "55857667604ffd4a1c632518208d81c8",
                "size": "2566645707",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/19fd8ae09ed64af1bc13a2e3aef32bec.zip"
            },
            "componentVersion": "2169.1.11.202201112316474729881"
        },
        {
            "componentId": "my_region_2169.11.17.202201112316474729879",
            "componentName": "my_region",
            "componentPackets": {
                "id": "foreign_my_region_2169.11.17.202201112316474729879_1_dc42916e2c87573cfd353de8b7b37ce2",
                "manualUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/9f53be97b5e34d95bac95607e876433c.zip",
                "md5": "dc42916e2c87573cfd353de8b7b37ce2",
                "size": "566531782",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/12/9f53be97b5e34d95bac95607e876433c.zip"
            },
            "componentVersion": "2169.11.17.202201112316474729879"
        }
    ],
    "decentralize": {
        "offset": 16687,
        "round": 28800,
        "strategyVersion": "8"
    },
    "description": {
        "firstTitle": "This release included Android security patch, fixed some known issues, and improved system stability.",
        "panelUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-2fffb28830a9a0cdfc987f20851180d8/component-ota/22/01/15/c6838b62ce1240a7a3294ec5782bce41.html",
        "share": ".",
        "url": "https://h5fsf.coloros.com/c05bb0389e8745cbb8482fcd71bf10a0/static/index.html#/about"
    },
    "descriptionType": 0,
    "googlePatchInfo": "0",
    "group": "RMX3370_11.A.07_0070_202201120154",
    "id": "61e60b0e66c96100cfa223a1",
    "isNvDescription": false,
    "isRecruit": false,
    "noticeType": 0,
    "nvId16": "NV1B",
    "osVersion": "ColorOS 11.2",
    "otaTemplateId": "61e2bee6124eec00cf837d40",
    "otaVersion": "RMX3370_11.A.07_0070_202201120154",
    "paramFlag": 1,
    "parent": "ota-template",
    "reminderType": 0,
    "rid": "c4e7771f-d204-4104-8ebf-cbda3a032377",
    "securityPatch": "2022-01-05",
    "securityPatchVendor": "2022-01-05",
    "silenceUpdate": 0,
    "status": "published",
    "timestampH5": "2022.01.15",
    "versionCode": 70,
    "versionName": "RMX3370_11_A.07",
    "versionTypeH5": "Official version"
}

R0rt1z2, 2022-01-18:
I guess you are looking for a full OTA link, which is unfortunately no longer available since Realme switched to components with RealmeUI 2.

mmetharam, 2022-01-18:
> I guess you are looking for a full OTA link, which is unfortunately no longer available since Realme switched to components with RealmeUI 2.

ah I see. I'm assuming there is no way to get (make?) a full OTA zip ourselves with all these components? Signature will be an issue.

R0rt1z2, 2022-01-18:
> > I guess you are looking for a full OTA link, which is unfortunately no longer available since Realme switched to components with RealmeUI 2.
> 
> ah I see. I'm assuming there is no way to get (make?) a full OTA zip ourselves with all these components? Signature will be an issue.

Not that I know, sorry.

mmetharam, 2022-01-18:
> Not that I know, sorry.

Noted, i'll just close this then as it's not really an issue with your script then. Thank you for such a quick explanation!

closed 2022-01-18 by mmetharam

macekboris, 2022-01-25:
Oh, it's just veird ... Reopening this thread for new comment regarding OTA's ...

On fully locked Realme 8 Pro (RMX3081) (GPRD EU) there is no way to get manual created ota's zip as you allready mentioned - Signature problem while making a OTA's zip .

For now only official flashing (OFR) via MSM in EDL is available because even partial components install via official Recovery is not working (nor full OFR zip rom via Recovery)

But veird while updating via update check is working, but since have a problem with access to data and temp download I cannot check the construct for successfull update.

No bootloader unlock/custom recovery/custom rom/root available for GDPR versions.

EXPORT versions have allready Android12 update available on RUI 3.0, for GDPR it is not available.