R0rt1z2/realme-ota #12: Please add support for Oppo phones.

100doses, 2022-02-14:
```
realme-ota CPH2219EEA CPH221920242_11.A.22_0220_202112040144
usage: realme-ota [-h] [-r {0,1,2,3}] [-d DUMP] [-o ONLY] [-s {0,1}] [-v {0,1}] product_model ota_version {1,2,3}
realme-ota: error: the following arguments are required: rui_version
```

R0rt1z2, 2022-02-18:
> I think i now understand for the RUI version, i'm on Android 11, so it's "2" right ? I just zipped and uploaded the OTA.apk file as attachment: [OTA.apk.zip](https://github.com/R0rt1z2/realme-ota/files/8087453/OTA.apk.zip)

Yeah that's correct, in your case RUI version should be `2`. I already tried to request it but it couldn't fetch anything.
Would you mind uploading the output of `adb shell getprop` here please (remove device-specific info, for instance S/N).

R0rt1z2, 2022-05-09:
Closing due to inactivity.

closed 2022-05-09 by R0rt1z2