R0rt1z2/realme-ota #15: The issue while using script

itsxrp, 2022-03-06:
Kindly help

python main.py RMX3361 RMX3361_11.A.01_001_202202101618 3 -r 1
Traceback (most recent call last):
  File "C:\Users\mnaum\Downloads\realme-ota-master\realme-ota-master\realme_ota\main.py", line 3, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'

R0rt1z2, 2022-03-06:
> Kindly help
> 
> python main.py RMX3361 RMX3361_11.A.01_001_202202101618 3 -r 1 Traceback (most recent call last): File "C:\Users\mnaum\Downloads\realme-ota-master\realme-ota-master\realme_ota\main.py", line 3, in import requests ModuleNotFoundError: No module named 'requests'

Install the "requests" pip package with `pip3 install requests`.
[I updated](b970d4dceefe489c87db3cdd5f24ecc0c4870c4f) the README to also include requests in the install process.

R0rt1z2, 2022-05-09:
Closing due to inactivity.

closed 2022-05-09 by R0rt1z2