R0rt1z2/realme-ota #16: F: Something went wrong while requesting to the endpoint :( Failed to parse: https://component-ota.coloros.com/update/v2!

AgafDL, 2022-03-15:
Hello! Everything worked gorgeous, after a couple of hours I again launched the script, but now only such an error:

python3 main.py RMX3461 RMX3461_11.A.05_0010_20220101111 2 -r 2

[2022-03-15 06:09:48.075190] I: Load payload for RMX3461 (RealmeUI V2)
[2022-03-15 06:09:48.076346] I: Wait for the endpoint to reply
[2022-03-15 06:09:48.077561] F: Something went wrong while requesting to the endpoint :( Failed to parse: https://component-ota-in.coloros.com/update/v1!

I read your topics, does not help.

AgafDL, 2022-03-16:
Hello. Please, help me. I need download full ota. I will attach the getprop file, there is data on the name of the OTA version. I can't download this version
[getprop_A03.txt](https://github.com/R0rt1z2/realme-ota/files/8262193/getprop_A03.txt)
.

AgafDL, 2022-03-16:
> Извините, я пропустил это. У меня вроде нормально работает?
> 
> ```
> [2022-03-16 12:52:31.705787] I: Wait for the endpoint to reply
> [2022-03-16 12:52:32.293182] I: All good
> [2022-03-16 12:52:32.293254] I: Let's rock
> {
>     "aid": "RMX3461NV1B_11.A",
>     "androidVersion": "Android 11",
>     "colorOSVersion": "ColorOS 11.2",
>     "componentAssembleType": true,
>     "components": [
>         {
>             "componentId": "21695_system_vendor_21695.1.5.202203040205035798237",
>             "componentName": "system_vendor",
>             "componentPackets": {
>                 "id": "foreign_21695_system_vendor_21695.1.5.202203040205035798237_1_3da0f8d53216da7dc972cd7bc0e48bfe",
>                 "manualUrl": "https://gauss-componentotacostmanual-in.allawnofs.com/remove-7fb0b8109410cd3f39195b062e67c26c/component-ota/22/03/04/3e4d374bee7a42ada5fc91cef85e7ad8.zip",
>                 "md5": "3da0f8d53216da7dc972cd7bc0e48bfe",
>                 "size": "4181403152",
>                 "type": "1",
>                 "url": "https://gauss-componentotacostmanual-in.allawnofs.com/remove-7fb0b8109410cd3f39195b062e67c26c/component-ota/22/03/04/3e4d374bee7a42ada5fc91cef85e7ad8.zip",
>                 "vabInfo": {
>                     "data": {
>                         "extra_params": "metadata_hash:af3ac36c60288e560260f72c4a9cc49a15f764d5561424edd05a701bbdcaacc2",
>                         "header": [
>                             "FILE_HASH=z6rhGKgO7jXHlox0KMP5XJNmkG/1AOugFnHBp+enMO8=",
>                             "FILE_SIZE=4181399254",
>                             "METADATA_HASH=DKADSD5quHZcMYzCSs1CFSglCMppdTNw9v2c92TAoQ0=",
>                             "METADATA_SIZE=207681",
>                             "ota_target_version=RMX3461_11.A.06_0060_202203040442"
>                         ],
>                         "otaStreamingProperty": "payload_metadata.bin:703:207948,payload.bin:703:4181399254,payload_properties.txt:4181400015:209,metadata:69:587",
>                         "vab_package_hash": "3da0f8d53216da7dc972cd7bc0e48bfe"
>                     }
>                 }
>             },
>             "componentVersion": "21695.1.5.202203040205035798237"
>         }
>     ],
>     "decentralize": {
>         "offset": 19902,
>         "round": 28800,
>         "strategyVersion": "default"
>     },
>     "description": {
>         "share": ".",
>         "url": "https://gauss-componentotacostmanual-in.allawnofs.com/remove-7fb0b8109410cd3f39195b062e67c26c/component-ota/22/03/14/f64becadddb8412db93f9e3ed5b5dfd7.html?logoType=1"
>     },
>     "googlePatchInfo": "0",
>     "group": "RMX3461_11.A.06_0060_202203040442",
>     "id": "622f084166c96100cfa3c401",
>     "isNvDescription": false,
>     "isRecruit": false,
>     "noticeType": 0,
>     "nvId16": "NV1B",
>     "osVersion": "ColorOS 11.2",
>     "otaTemplateId": "622efdc8124eec00cf8529f7",
>     "otaVersion": "RMX3461_11.A.06_0060_202203040442",
>     "paramFlag": 1,
>     "parent": "ota-template",
>     "reminderType": 0,
>     "rid": "d1059985-975b-4720-a6ab-60bc8f77ae9e",
>     "securityPatch": "2022-03-05",
>     "securityPatchVendor": "2022-03-05",
>     "silenceUpdate": 0,
>     "status": "published",
>     "timestampH5": "2022.03.14",
>     "versionCode": 60,
>     "versionName": "RMX3461_11_A.06",
>     "versionTypeH5": "Official version"
> }
> ```

Yes, I reinstalled and worked fine, thanks. Can you answer my second question?

AgafDL, 2022-03-17:
How to download this version?
[image]

AgafDL, 2022-05-06:
Hello, please, help with this model and firmware. There is a Realme Q3S phone(rmx3461) with Chinese and English firmware (for the Kenya region). For the Chinese version, with the help of your script, the firmware is easily located, for the Kenyan firmware - no. Perhaps there is a server for Kenyan, African firmware? Could you help, please? I attached the Getprop file from the Kenya firmware.
[getprop_A03.txt](https://github.com/R0rt1z2/realme-ota/files/8642000/getprop_A03.txt)

R0rt1z2, 2022-05-09:
Duplicate of https://github.com/R0rt1z2/realme-ota/issues/20.

closed 2022-05-09 by R0rt1z2