R0rt1z2/realme-ota #20: Hello, please, help with this model and firmware. Realme q3s(rmx3461). Maybe need add african server Realme?

AgafDL, 2022-05-06:
Hello, please, help with this model and firmware. There is a Realme Q3S phone(rmx3461) with Chinese and English firmware (for the Kenya region). For the Chinese version, with the help of your script, the firmware is easily located, for the Kenyan firmware - no. Perhaps there is a server for Kenyan, African firmware? Could you help, please? I attached the Getprop file from the Kenya firmware.

[getprop_A03.txt](https://github.com/R0rt1z2/realme-ota/files/8642013/getprop_A03.txt)

R0rt1z2, 2022-05-08:
Can you try to pull `/my_region/etc/extension/appfeature.region_in.android_coreservices.xml`?

R0rt1z2, 2022-05-08:
EDIT: managed to fetch the OTA, will update the tool as soon as possible.
```{
    "aid": "RMX3461T2NV74_11.A",
    "androidVersion": "Android 11",
    "colorOSVersion": "ColorOS 11.1",
    "componentAssembleType": true,
    "components": [
        {
            "componentId": "21643_system_vendor_21643.2.1.202204240207287039083",
            "componentName": "system_vendor",
            "componentPackets": {
                "id": "foreign_21643_system_vendor_21643.2.1.202204240207287039083_1_67007a3c65d509314823e7894f03cafa",
                "manualUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-881136a9c6a1a423d27f9d1c142b6ecf/component-ota/22/04/29/db686713d5374a0abcea7a1a50c2bb34.zip",
                "md5": "67007a3c65d509314823e7894f03cafa",
                "size": "3977926661",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-881136a9c6a1a423d27f9d1c142b6ecf/component-ota/22/04/29/db686713d5374a0abcea7a1a50c2bb34.zip",
                "vabInfo": {
                    "data": {
                        "extra_params": "metadata_hash:0b92512f238a54eed6f91137fc29b4e9c3a4b152c5bf94dfe72178f0c9c4a1f4",
                        "header": [
                            "FILE_HASH=v1fFmePmDfMoqe/IlCvwqzXGJaGOmoXKQrsSximDurs=",
                            "FILE_SIZE=3977922753",
                            "METADATA_HASH=SIMOf62lwuz7BJ2IOZxaPWa1/QYaBnM41ZH49pnEUeY=",
                            "METADATA_SIZE=198156",
                            "ota_target_version=RMX3461T2_11.A.06_0060_202204240423"
                        ],
                        "otaStreamingProperty": "payload_metadata.bin:711:198423,payload.bin:711:3977922753,payload_properties.txt:3977923522:211,metadata:69:595",
                        "vab_package_hash": "67007a3c65d509314823e7894f03cafa"
                    }
                }
            },
            "componentVersion": "21643.2.1.202204240207287039083"
        }
    ],
    "decentralize": {
        "offset": 20424,
        "round": 28800,
        "strategyVersion": "default"
    },
    "description": {
        "firstTitle": "This update integrates an Android security patch, fixes some known issues, and improves system stability.",
        "panelUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-881136a9c6a1a423d27f9d1c142b6ecf/component-ota/22/05/06/a5623145e13849d6806541bfc4ce4ccb.html",
        "share": ".",
        "url": "https://h5fsf.coloros.com/37aa9ab5f0784f509b1e5394fb6ecda8/static/index.html#/about"
    },
    "descriptionType": 0,
    "googlePatchInfo": "0",
    "group": "RMX3461T2_11.A.06_0060_202204240423",
    "id": "626b536166c961010889b5a4",
    "isNvDescription": false,
    "isRecruit": false,
    "noticeType": 0,
    "nvId16": "NV74",
    "osVersion": "ColorOS 11.1",
    "otaTemplateId": "627490d466c961010889cd87",
    "otaVersion": "RMX3461T2_11.A.06_0060_202204240423",
    "paramFlag": 1,
    "parent": "ota-template",
    "reminderType": 0,
    "rid": "36343e5e-4d70-40ca-93b9-e0c94538431f",
    "securityPatch": "2022-04-05",
    "securityPatchVendor": "2022-04-05",
    "silenceUpdate": 0,
    "status": "published",
    "timestampH5": "2022.05.06",
    "versionCode": 60,
    "versionName": "RMX3461T2_11_A.06",
    "versionTypeH5": "Official version"
}
```

closed 2022-05-08 by R0rt1z2