R0rt1z2/realme-ota #21: Response contents mismatch, expected 'body' got 'artifactV1Result is empty'!

Ma8uru77, 2022-05-16:
I did everything as told.

R0rt1z2, 2022-05-16:
> I did everything as told.
> 
> This is what I wrote and the response I got.
> 
> root@localhost:/usr/local/lib/python3.9/dist-packages/realme_ota# python3 main.py RMX3561 RMX3561_11.A.06_0060_202205061337 3 00011011 [2022-05-16 16:59:19.952202] I: Load payload for RMX3561 (RealmeUI V3) [2022-05-16 16:59:19.953314] I: Wait for the endpoint to reply [2022-05-16 16:59:20.149779] F: Response contents mismatch, expected 'body' got 'artifactV1Result is empty'!

Interesting, what phone is this?

Ma8uru77, 2022-05-16:
[props.txt](https://github.com/R0rt1z2/realme-ota/files/8702263/props.txt)

GT neo 3. I tried both the chinese variant (mine) and the indian variant (Got the props file for someone on telegram).

python3 main.py RMX3561 RMX3561_11.A.06_0060_202205061337 3 00011011
python3 main.py RMX3562 RMX3562_11.A.27_0270_202204292232 3 10010111

Both didn't work.

Ma8uru77, 2022-05-16:
Tried different regions with -r and changing the last 16 digits to 0. Still didn't work.

aScordino, 2022-05-21:
Hey,

I am unsure about this, but my guess is that realme did something for **RMX3562** that makes it unusable with this tool. I've tried everything, even reverting to an older commit, but still did nothing. I really don't know, nor Roger does. It works fine on **RMX3370** (_realme GT Neo2_).

I've even tried to replace `RMX3561_11.A.06_0060_202205061337` with `RMX3561_11.A.06_0000_000000000000`, which still did not work (further explaination will be added by Roger soon).

But, before giving up, could you try sending an APK for the OTA? 
```bash
$ adb shell find /system/ | grep OTA
$ adb pulll /system/<FILE> /path/to/wherever/you/want/to/pull/this
```
Or you can use any file explorer (e.g: MiXplorer) to directly share it here or on Telegram (to me, [**@aScordino**](https://t.me/aScordino) or to Roger, [**@R0rt1z2**](https://t.me/R0rt1z2)).

Good luck,
Antonino.

AgafDL, 2022-06-24:
Hello. I have a similar problem. Earlier, your script was correctly looking for updates for the Chinese Realme Q3S firmware. Now he does not find Ota update for Android 11 and 12. The same error as the author. I will attach GetProp.
NV_ID for Realme Q3s:
China region- 10010111
Kenya region- 01110100
India region- 00011011

[getprop_Realme_Q3s_RUI3_Android12.txt](https://github.com/R0rt1z2/realme-ota/files/8977548/getprop_Realme_Q3s_RUI3_Android12.txt)

R0rt1z2, 2022-06-24:
> Hello. I have a similar problem. Earlier, your script was correctly looking for updates for the Realme Q3S with chinese firmware. Now he does not find Ota update for Android 11 and 12. The same error as the author. I will attach GetProp. NV_ID for Realme Q3s: China region- 10010111 Kenya region- 01110100 India region- 00011011
> 
> [getprop_Realme_Q3s_RUI3_Android12.txt](https://github.com/R0rt1z2/realme-ota/files/8977548/getprop_Realme_Q3s_RUI3_Android12.txt)

`realme-ota RMX3461 RMX3461_11.C.01_0000_000000000000 3 10010111 -r 1` works fine for me:
```json
{
    "aid": "RMX3461NV97_11.C",
    "androidVersion": "Android 12",
    "colorOSVersion": "ColorOS 12.0",
    "componentAssembleType": true,
    "components": [
        {
            "componentId": "my_manifest_RMX3461_11.C.03_1030_202204130507.97.da6f1ade",
            "componentName": "my_manifest",
            "componentPackets": {
                "id": "domestic_my_manifest_RMX3461_11.C.03_1030_202204130507.97.da6f1ade_1_d086ae3a164ebf5f75b52edacf3bc666",
                "manualUrl": "https://gauss-componentotacostmanual-cn.allawnfs.com/remove-cfa0ba9ec2636d24ed6d50dbcf96912b/component-ota/22/04/15/e56c559eb9f0416d94d885debc645d16.zip",
                "md5": "d086ae3a164ebf5f75b52edacf3bc666",
                "size": "5097936838",
                "type": "1",
                "url": "https://gauss-compotacostauto-cn.allawnfs.com/remove-cfa0ba9ec2636d24ed6d50dbcf96912b/component-ota/22/04/15/e56c559eb9f0416d94d885debc645d16.zip",
                "vabInfo": {
                    "data": {
                        "extra_params": "metadata_hash:3dc65ff774c906a168a8819cc10d03d6ed07e1bff88a883f32fe34b231e18289",
                        "header": [
                            "FILE_HASH=T+oLbPPLPtbNA+D3v2Dk/Omac8DZwHi7GT92oVJukm8=",
                            "FILE_SIZE=5097931161",
                            "METADATA_HASH=cfF6/sRCHTydsyRkQwbbK+wEJZawAVkxFmGmrF50hDE=",
                            "METADATA_SIZE=197119",
                            "ota_target_version=RMX3461_11.C.03_1030_202204130507"
                        ],
                        "otaStreamingProperty": "payload_metadata.bin:2278:197386,payload.bin:2278:5097931161,payload_properties.txt:5097933497:209,metadata:69:861,metadata.pb:998:1213",
                        "vab_package_hash": "d086ae3a164ebf5f75b52edacf3bc666"
                    }
                }
            },
            "componentVersion": "RMX3461_11.C.03_1030_202204130507.97.da6f1ade"
        }
    ],
    "decentralize": {
        "offset": 12664,
        "round": 28800,
        "strategyVersion": "8"
    },
    "description": {
        "firstTitle": "realme UI \u7248\u672c\u5347\u7ea7\u4e3a realme UI 3.0\uff0cAndroid \u7248\u672c\u5347\u7ea7\u5230 Android 12\u3002\n\u4e3a\u4f60\u5e26\u6765\u66f4\u4e2a\u6027\u3001\u66f4\u9ad8\u6548\u3001\u66f4\u7545\u5feb\u7684\u7545\u901f\u6f6e\u73a9\u4f53\u9a8c\uff01",
        "panelUrl": "https://gauss-compotacostauto-cn.allawnfs.com/remove-cfa0ba9ec2636d24ed6d50dbcf96912b/component-ota/22/04/21/8398f4a59dd840969c956f5874faa233.html",
        "share": ".",
        "url": "https://h5fs.coloros.com/56200f0bde584a80a9051f83e01a6301/static/index.html#/about"
    },
    "descriptionType": 0,
    "googlePatchInfo": "0",
    "group": "RMX3461_11.C.03_1030_202204130507",
    "id": "6259186a66c9610108891b79",
    "isNvDescription": false,
    "isRecruit": false,
    "noticeType": 0,
    "nvId16": "NV97",
    "osVersion": "ColorOS 12.0",
    "otaTemplateId": "62562ac9124eec010d7899ef",
    "otaVersion": "RMX3461_11.C.03_1030_202204130507",
    "paramFlag": 1,
    "parent": "ota-template",
    "reminderType": 0,
    "rid": "19b6b253-8871-4f5f-bd51-9a29a29947d4",
    "securityPatch": "2022-04-05",
    "securityPatchVendor": "2022-04-05",
    "silenceUpdate": 0,
    "status": "published",
    "timestampH5": "2022.04.21",
    "versionCode": 1030,
    "versionName": "RMX3461_11_C.03",
    "versionTypeH5": "\u6b63\u5f0f\u7248"
}
```

AgafDL, 2022-06-24:
> > Привет. У меня аналогичная проблема. Ранее ваш скрипт корректно искал обновления для Realme Q3S с китайской прошивкой. Теперь не находит Ота обновления на Андроид 11 и 12. Та же ошибка что и у автора. Я прикреплю GetProp. NV_ID для Realme Q3s: регион Китая — 10010111 регион Кении — 01110100 регион Индии — 00011011
> > [getprop_Realme_Q3s_RUI3_Android12.txt](https://github.com/R0rt1z2/realme-ota/files/8977548/getprop_Realme_Q3s_RUI3_Android12.txt)
> 
> `realme-ota RMX3461 RMX3461_11.C.01_0000_000000000000 3 10010111 -r 1`отлично работает для меня:
> 
> ```json
> {
>     "aid": "RMX3461NV97_11.C",
>     "androidVersion": "Android 12",
>     "colorOSVersion": "ColorOS 12.0",
>     "componentAssembleType": true,
>     "components": [
>         {
>             "componentId": "my_manifest_RMX3461_11.C.03_1030_202204130507.97.da6f1ade",
>             "componentName": "my_manifest",
>             "componentPackets": {
>                 "id": "domestic_my_manifest_RMX3461_11.C.03_1030_202204130507.97.da6f1ade_1_d086ae3a164ebf5f75b52edacf3bc666",
>                 "manualUrl": "https://gauss-componentotacostmanual-cn.allawnfs.com/remove-cfa0ba9ec2636d24ed6d50dbcf96912b/component-ota/22/04/15/e56c559eb9f0416d94d885debc645d16.zip",
>                 "md5": "d086ae3a164ebf5f75b52edacf3bc666",
>                 "size": "5097936838",
>                 "type": "1",
>                 "url": "https://gauss-compotacostauto-cn.allawnfs.com/remove-cfa0ba9ec2636d24ed6d50dbcf96912b/component-ota/22/04/15/e56c559eb9f0416d94d885debc645d16.zip",
>                 "vabInfo": {
>                     "data": {
>                         "extra_params": "metadata_hash:3dc65ff774c906a168a8819cc10d03d6ed07e1bff88a883f32fe34b231e18289",
>                         "header": [
>                             "FILE_HASH=T+oLbPPLPtbNA+D3v2Dk/Omac8DZwHi7GT92oVJukm8=",
>                             "FILE_SIZE=5097931161",
>                             "METADATA_HASH=cfF6/sRCHTydsyRkQwbbK+wEJZawAVkxFmGmrF50hDE=",
>                             "METADATA_SIZE=197119",
>                             "ota_target_version=RMX3461_11.C.03_1030_202204130507"
>                         ],
>                         "otaStreamingProperty": "payload_metadata.bin:2278:197386,payload.bin:2278:5097931161,payload_properties.txt:5097933497:209,metadata:69:861,metadata.pb:998:1213",
>                         "vab_package_hash": "d086ae3a164ebf5f75b52edacf3bc666"
>                     }
>                 }
>             },
>             "componentVersion": "RMX3461_11.C.03_1030_202204130507.97.da6f1ade"
>         }
>     ],
>     "decentralize": {
>         "offset": 12664,
>         "round": 28800,
>         "strategyVersion": "8"
>     },
>     "description": {
>         "firstTitle": "realme UI \u7248\u672c\u5347\u7ea7\u4e3a realme UI 3.0\uff0cAndroid \u7248\u672c\u5347\u7ea7\u5230 Android 12\u3002\n\u4e3a\u4f60\u5e26\u6765\u66f4\u4e2a\u6027\u3001\u66f4\u9ad8\u6548\u3001\u66f4\u7545\u5feb\u7684\u7545\u901f\u6f6e\u73a9\u4f53\u9a8c\uff01",
>         "panelUrl": "https://gauss-compotacostauto-cn.allawnfs.com/remove-cfa0ba9ec2636d24ed6d50dbcf96912b/component-ota/22/04/21/8398f4a59dd840969c956f5874faa233.html",
>         "share": ".",
>         "url": "https://h5fs.coloros.com/56200f0bde584a80a9051f83e01a6301/static/index.html#/about"
>     },
>     "descriptionType": 0,
>     "googlePatchInfo": "0",
>     "group": "RMX3461_11.C.03_1030_202204130507",
>     "id": "6259186a66c9610108891b79",
>     "isNvDescription": false,
>     "isRecruit": false,
>     "noticeType": 0,
>     "nvId16": "NV97",
>     "osVersion": "ColorOS 12.0",
>     "otaTemplateId": "62562ac9124eec010d7899ef",
>     "otaVersion": "RMX3461_11.C.03_1030_202204130507",
>     "paramFlag": 1,
>     "parent": "ota-template",
>     "reminderType": 0,
>     "rid": "19b6b253-8871-4f5f-bd51-9a29a29947d4",
>     "securityPatch": "2022-04-05",
>     "securityPatchVendor": "2022-04-05",
>     "silenceUpdate": 0,
>     "status": "published",
>     "timestampH5": "2022.04.21",
>     "versionCode": 1030,
>     "versionName": "RMX3461_11_C.03",
>     "versionTypeH5": "\u6b63\u5f0f\u7248"
> }
> ```
Many thanks for the help. )

AgafDL, 2022-06-24:
Interesting. Version C04 is out, but your script only finds C03. This version does not appear immediately, do we have to wait? People are already downloading the update.

[image]

Topdog3518, 2022-07-28:
It resulted in success in termux but end result in out.txt was fail. Please help.
[image]
[image]

R0rt1z2, 2022-07-28:
> It resulted in success in termux but end result in out.txt was fail. Please help. [image]

Can you please show the arguments you used to fetch the OTA?

Topdog3518, 2022-07-28:
[image]

Topdog3518, 2022-07-28:
> 

[image]

Topdog3518, 2022-07-28:
> > It resulted in success in termux but end result in out.txt was fail. Please help.
> 
> Can you please show the arguments you used to fetch the OTA?

getprop ro.build.version.ota

Topdog3518, 2022-07-28:
[image]

R0rt1z2, 2022-08-05:
@Topdog3518 can you please send a full getprop?

Topdog3518, 2022-08-05:
> @Topdog3518 can you please send a full getprop?

[prop.txt](https://github.com/R0rt1z2/realme-ota/files/9268673/prop.txt)

Topdog3518, 2022-08-11:
How to add various components to one .ozip file. Or how to flash them separately ?

masd0sr, 2022-09-22:
it's problem((

$ realme-ota RMX3561 RMX3561_11.A.09_0090_2022206042216 3 10100110
[2022-09-22 17:53:31.465260] I: Load payload for RMX3561 (RealmeUI V3)
[2022-09-22 17:53:31.467942] I: Wait for the endpoint to reply
[2022-09-22 17:53:33.126834] I: Load payload for RMX3561 (RealmeUI V3)
[2022-09-22 17:53:33.129524] I: Wait for the endpoint to reply
[2022-09-22 17:53:34.486030] F: Response status mismatch, expected '200' got '304' (artifactV1Result is empty)!

hanajoruno, 2022-10-12:
save artifactV1Result problem, maybe should to modify the request body information, like romVersion, imei, deviceid

hanajoruno, 2022-10-13:
In short, the appearance of artifactV1Result means that the current device does not have permission to obtain ota. I made the following attempts:
My LE2120 has applied for OBT and has no CBT permission. I can get OBT F.11 by using the same request data under CBT F.12. I only need to change the romVersion and otaVersion in the request headers.
For the same request data, changing the deviceid in the headers to obtain CBT F.12 does not work, indicating that there are other information in the data that have not changed.
It is recommended that more custom parameters be added.

R0rt1z2, 2022-10-17:
Closing (as well) as per https://github.com/R0rt1z2/realme-ota/pull/31.

closed 2022-10-17 by R0rt1z2