R0rt1z2/realme-ota #37: Unable to install realme-ota on windows

aruto768, 2022-12-06:
[image]

Using Terminal run as admin, new tab PowerShell put this

# (Requires privilegies)
Invoke-WebRequest https://cdn.r0rt1z2.com/realme-ota/Install.ps1 | Invoke-Expression

i got error, like that

JunMuraKami, 2022-12-08:
I'm facing the same issue.
The URL doesn't seem to exist.

`Invoke-WebRequest -Uri https://gist.githubusercontent.com/R0rt1z2/9b4600b5bb11d0750bdcebea04374c63/raw/b4df9d6ff725904b2115bb8353bd1fcb6de24b5b/realme-ota.bat -Outfile realme-ota.bat`

R0rt1z2, 2022-12-08:
The wrapper wasn't being downloaded at all, it should work now, please try again.

closed 2022-12-08 by R0rt1z2