R0rt1z2/realme-ota #42: Response status mismatch, expected '200' got '304' (artifactV1Result is empty)!

v2t9484, 2023-01-14:
C:\Windows\system32>realme-ota RMX3371 RMX3371_11.C.00_0000_000000000000 3 00111100
[2023-01-14 11:11:27.708970] I: Load payload for RMX3371 (RealmeUI V3)
[2023-01-14 11:11:27.709771] I: Wait for the endpoint to reply
[2023-01-14 11:11:27.968881] F: Response status mismatch, expected '200' got '304' (artifactV1Result is empty)!
Press any key to continue . . .

amigaser, 2023-01-14:
The same problem with other device (EU region):
```
f:\Realme\Update\realme-ota-master\realme_ota>main.py RMX3393 RMX3393_11.C.05_0000_000000000000 4 01000100
[2023-01-14 22:23:01.262213] I: Load payload for RMX3393 (RealmeUI V4)
[2023-01-14 22:23:01.262213] I: Wait for the endpoint to reply
[2023-01-14 22:23:02.193228] F: Response status mismatch, expected '200' got '304' (artifactV1Result is empty)!
```
On the device update comes.

R0rt1z2, 2023-01-15:
> The same problem with other device (EU region):
>
> ```
> f:\Realme\Update\realme-ota-master\realme_ota>main.py RMX3393 RMX3393_11.C.05_0000_000000000000 4 01000100
> [2023-01-14 22:23:01.262213] I: Load [...]

Next time try to use '0' as the last parameter (nvId):

```json

{
    "aid": "RMX3371NV1B_11.C",
    "androidVersion": "Android 13",
    "colorOSVersion": "ColorOS 13.0",
    "componentAssembleType": false,
    "components": [
        {
            "componentId": "my_manifest_RMX3371_11.C.04_1040_202212232109.1B.546554dc",
            "componentName": "my_manifest",
            "componentPackets": {
                "id": "foreign_my_manifest_RMX3371_11.C.04_1040_202212232109.1B.546554dc_1_4bdce6b96058abf6e8853c61064b6248",
                "manualUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-edcd4152d6632a7cb942026d1e1347a5/component-ota/22/12/24/b5f0915e923b4e1c8aec9c72ebaefc62.zip",
                "md5": "4bdce6b96058abf6e8853c61064b6248",
                "size": "5340388494",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-edcd4152d6632a7cb942026d1e1347a5/component-ota/22/12/24/b5f0915e923b4e1c8aec9c72ebaefc62.zip",
                "vabInfo": {
                    "data": {
                        "extra_params": "metadata_hash:c63bca3e03165b03348fd850906acc8fe000ac4abd887f26073ead6604c76974",
                        "header": [
                            "FILE_HASH=aDpMI63Rs92IO6fLZCDKfsHkKziXknjBZ+gmoOH67Bw=",
                            "FILE_SIZE=5340382806",
                            "METADATA_HASH=jnCfRF6F6MydyLlUL1AoKEZDRNQFhCnSg3sRfxn8opI=",
                            "METADATA_SIZE=207605",
                            "oplus_update_engine_verify_disable=0",
                            "oplus_rom_version=V13.0.0",
                            "security_patch=2022-12-05",
                            "oplus_separate_soft=21732",
                            "security_patch_vendor=2022-12-05",
                            "ota_target_version=RMX3371_11.C.04_1040_202212232109"
                        ],
                        "otaStreamingProperty": "payload_metadata.bin:2141:207872,payload.bin:2141:5340382806,payload_properties.txt:5340385005:357,metadata:69:989,metadata.pb:1126:948",
                        "vab_package_hash": "4bdce6b96058abf6e8853c61064b6248"
                    }
                }
            },
            "componentVersion": "RMX3371_11.C.04_1040_202212232109.1B.546554dc"
        }
    ],
    "decentralize": {
        "offset": 11477,
        "round": 28800,
        "strategyVersion": "default"
    },
    "description": {
        "firstTitle": "A system update is available. Your system will be upgraded to Android 13. A new design philosophy, Aquamorphic Design, is adopted for this update. Highlights include vibrant theme colours, optimised task management tools, enhanced security and privacy, new Always-On Display styles and refined visual effects. ",
        "panelUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-edcd4152d6632a7cb942026d1e1347a5/component-ota/22/12/28/5186e08b40dd4bcda88fca9982500205.html",
        "share": ".",
        "url": "https://h5fsf.coloros.com/3adb669b49144103b5afb789182233ec/static/index.html#/about"
    },
    "googlePatchInfo": "0",
    "group": "RMX3371_11.C.04_1040_202212232109",
    "id": "63a6cadc124eec010e6e949c",
    "isNvDescription": false,
    "isRecruit": false,
    "isSecret": false,
    "noticeType": 0,
    "nvId16": "NV1B",
    "osVersion": "ColorOS 13.0",
    "otaTemplateId": "63a6cadc66c961010e2f5838",
    "otaVersion": "RMX3371_11.C.04_1040_202212232109",
    "paramFlag": 1,
    "parent": "ota-template",
    "realAndroidVersion": "Android 13",
    "realOsVersion": "ColorOS 13.0",
    "reminderType": 0,
    "rid": "37d02511-0aa1-4150-bb5e-9534dad6291c",
    "securityPatch": "2022-12-05",
    "securityPatchVendor": "2022-12-05",
    "silenceUpdate": 0,
    "status": "published",
    "timestampH5": "2022.12.28",
    "versionCode": 1040,
    "versionName": "RMX3371_11_C.04",
    "versionTypeH5": "Official version"
}
```

R0rt1z2, 2023-01-15:
```json

{
    "aid": "RMX3393NV1B_11.C",
    "androidVersion": "Android 13",
    "colorOSVersion": "ColorOS 13.0",
    "componentAssembleType": true,
    "components": [
        {
            "componentId": "my_manifest_RMX3393_11.C.06_1060_202301111732.1B.4fb928a3",
            "componentName": "my_manifest",
            "componentPackets": {
                "id": "foreign_my_manifest_RMX3393_11.C.06_1060_202301111732.1B.4fb928a3_1_c0ea55674d7ab41675875b88f96671bf",
                "manualUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-a9539f3b94c4f76fbb2d9babaafe179c/component-ota/23/01/12/4bada651bb9948deb4d65c66c0cf504b.zip",
                "md5": "c0ea55674d7ab41675875b88f96671bf",
                "size": "5365587212",
                "type": "1",
                "url": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-a9539f3b94c4f76fbb2d9babaafe179c/component-ota/23/01/12/4bada651bb9948deb4d65c66c0cf504b.zip",
                "vabInfo": {
                    "data": {
                        "extra_params": "metadata_hash:c023b7bd4e680e3eed6b8a029ec78a85da99e27059ee66d7675e534b2c95bfcc",
                        "header": [
                            "FILE_HASH=0lRCVzVdnUNov7c7ctnkrhZ3dNC9qiSHQfjUqSpUZOI=",
                            "FILE_SIZE=5365581565",
                            "METADATA_HASH=xaAudKdzqBrbyyQGem1V0dLrD7k+Qdb1lijIJPoRpxQ=",
                            "METADATA_SIZE=201661",
                            "oplus_separate_soft=216CA",
                            "oplus_rom_version=V13.0.0",
                            "oplus_update_engine_verify_disable=0",
                            "security_patch_vendor=2022-12-05",
                            "security_patch=2022-12-05",
                            "ota_target_version=RMX3393_11.C.06_1060_202301111732"
                        ],
                        "otaStreamingProperty": "payload_metadata.bin:2100:201928,payload.bin:2100:5365581565,payload_properties.txt:5365583723:357,metadata:69:979,metadata.pb:1116:917",
                        "vab_package_hash": "c0ea55674d7ab41675875b88f96671bf"
                    }
                }
            },
            "componentVersion": "RMX3393_11.C.06_1060_202301111732.1B.4fb928a3"
        }
    ],
    "decentralize": {
        "offset": 5686,
        "round": 28800,
        "strategyVersion": "default"
    },
    "description": {
        "firstTitle": "This update integrates the latest Android security patches, fixes some known issues, and improves system performance.",
        "panelUrl": "https://gauss-componentotacostmanual-sg.allawnofs.com/remove-a9539f3b94c4f76fbb2d9babaafe179c/component-ota/23/01/12/cf29bf6384bc44de82aa27049b5f8697.html",
        "share": ".",
        "url": "https://h5fsf.coloros.com/3adb669b49144103b5afb789182233ec/static/index.html#/about"
    },
    "googlePatchInfo": "0",
    "group": "RMX3393_11.C.06_1060_202301111732",
    "id": "63bf696a124eec010ea95290",
    "isNvDescription": false,
    "isRecruit": false,
    "isSecret": false,
    "noticeType": 0,
    "nvId16": "NV1B",
    "osVersion": "ColorOS 13.0",
    "otaTemplateId": "63bf696a66c961010ecb9d86",
    "otaVersion": "RMX3393_11.C.06_1060_202301111732",
    "paramFlag": 1,
    "parent": "ota-template",
    "realAndroidVersion": "Android 13",
    "realOsVersion": "ColorOS 13.0",
    "reminderType": 0,
    "rid": "6c5b4203-8ffb-4d09-aa3f-c1c87fbd181a",
    "securityPatch": "2022-12-05",
    "securityPatchVendor": "2022-12-05",
    "silenceUpdate": 0,
    "status": "published",
    "timestampH5": "2023.01.12",
    "versionCode": 1060,
    "versionName": "RMX3393_11_C.06",
    "versionTypeH5": "Official versions"
}
```

closed 2023-01-15 by R0rt1z2

amigaser, 2023-01-15:
This is for the Indian region (NV1B), not for Europe. I need OTA for NV44 (01000100). Is it not on the server?

rapperskull, 2023-01-15:
Many users are not receiving the update on their phones (at least I've heard reports about the 9 Pro Plus). Maybe the rollout has been halted or it's still limited to some countries. On a real phone, using a VPN sometimes helps in finding the update. You can try it with realme-ota too.

amigaser, 2023-01-17:
> You can try it with realme-ota too.
With VPN still does not find update for NV44.

PrinceMadds, 2025-05-03:
> {
>     "aid": "RMX3393NV1B_11.C",
>     "androidVersion": "Android 13",
>     "colorOSVersion": "ColorOS 13.0",
>     "componentAssembleType": true,
>     "components": [
>         {
>             "componentId": [...]

hello please help me, I have tried it in termux but it only says Response status mismatch, expected '200' got '304' (artifactV1Result is empty)!

no build RMX3624_13.C.36_1360_20241017091737

amigaser, 2025-05-03:
You cannot get OTA firmware through this script for your model. There are only service firmware for the SPD Flash Tool for it.

PrinceMadds, 2025-05-03:
> Anda tidak bisa mendapatkan firmware OTA melalui skrip ini untuk model Anda. Hanya ada firmware layanan untuk SPD Flash Tool untuk model tersebut.

but my friend can get the download link for the model

amigaser, 2025-05-03:
Give the command that your friend was able to get the OTA firmware through this script. I can't get it.

amigaser, 2025-05-04:
I found a command for your phone. You can only get the latest update.
main.py -r 0 RMX3624 RMX3624_13.C.00_0000_000000000000 1 0

PrinceMadds, 2025-05-04:
> Saya menemukan perintah untuk telepon Anda. Anda hanya bisa mendapatkan pembaruan terkini. main.py -r 0 RMX3624 RMX3624_13.C.00_0000_000000000000 1 0

thanks bro,