R0rt1z2/realme-ota #50: Unable to fetch RMX3572 EU OTA

alibei, 2023-05-22:
Have problems to fetch the rui 4.0 eu ota for RMX3572. 
The error indicates "Response status mismatch, expected '200' got '304' (artifactV1Result is empty)!"

ro.product.name: RMX3572EEA
ro.build.version.ota: RMX3572_11.C.03_1030_202302081756
ro.build.version.realmeui: V4.0
ro.build.oplus_nv_id: 01000100

Tried the following commands:
realme-ota RMX3572EEA RMX3572_11.C.03_1030_202302081756 4 01000100
realme-ota RMX3572EEA RMX3572_11.C.03_1030_202302081756 4 0
realme-ota RMX3572 RMX3572_11.C.03_1030_202302081756 4 01000100
realme-ota RMX3572 RMX3572_11.C.03_1030_202302081756 4 0

Fetching a RUI 3 build worked:
realme-ota RMX3572EEA RMX3572_11.A.06_2022_062212210000 3 01000100
realme-ota RMX3572 RMX3572_11.A.06_2022_062212210000 3 01000100

rapperskull, 2023-05-24:
You have to add `-r 3` to use the EU server. This works fine:
`realme-ota -r 3 RMX3572EEA RMX3572_11.C.03_1030_202302081756 4 01000100`

alibei, 2023-05-24:
I used a proxy from the USA, there it is successful but it gives me the C.04 build which is for another region.
Was now able to get the link another way from a friend
https://rms01.realme.net/sw/RMX3572GDPR_11_C.03_2023020817560000.zip

alibei, 2023-05-24:
Yeah my bad was confused for a little. I edited my post before because I saw my mistake