R0rt1z2/realme-ota #55: F: Response status mismatch, expected '200' got '203' (otaVersion parse exception)!

chackAJMCPE, 2023-08-30:
Idk what to do. Please, help.

CallistoAlex, 2023-09-03:
You inputted incorrect parameters.

Your ota_version value inputted is incorrect.

RMX3472_11.C.08_1080_202212122313 is the correct example. 
You may replace each digit in 1080_202212122313 with 0 if you don't know yours.

Moreover, you forgot to indicate the nv_id. My example below is 111001 referring to Thailand. If you don't know it, you can indicate it as 0, as explained in the manual of the script.

Here is my working command, for your information.

python3 main.py -r 0 RMX3472 RMX3472_11.C.08_1080_202212122313 4 111001

chackAJMCPE, 2023-09-04:
So i need for global (eu)

chackAJMCPE, 2023-09-04:
Nevermind. I figured it out.

bnee99, 2023-09-30:
> Nevermind. I figured it out.

can you provide your input. I'm trying to download update file for RMX3063 and i'm on custom rom.
ota version is RMX3063_11_A.68

closed 2023-12-04 by R0rt1z2