R0rt1z2/realme-ota #62: The script output differs from the actually available version

Jipok, 2024-01-19:
I have:
[image]
So i used command:
`realme-ota RMX3560 RMX3560_11.C.27_1270_202312042122 4 10010111 -r 1`
<details><summary>Output</summary>

``` 
{
    "aid": "RMX3560NV97_11.C",
    "androidVersion": "Android 13",
    "colorOSVersion": "ColorOS 13.1",
    "componentAssembleType": true,
    "components": [
        {
            "componentId": "my_manifest_RMX3560_11.C.27_1270_202312042122.97.27ae529a",
            "componentName": "my_manifest",
            "componentPackets": {
                "id": "domestic_my_manifest_RMX3560_11.C.27_1270_202312042122.97.27ae529a_1_3797b83e14892813c93b860251e18e2a",
                "manualUrl": "https://gauss-componentotacostmanual-cn.allawnfs.com/remove-441ae418e0930b26531f1e13ac7d638b/component-ota/23/12/08/c5a9de5aa6a146f3babf3ea0ad757f64.zip",
                "md5": "3797b83e14892813c93b860251e18e2a",
                "size": "7320952141",
                "type": "1",
                "url": "https://gauss-compotacostauto-cn.allawnfs.com/remove-441ae418e0930b26531f1e13ac7d638b/component-ota/23/12/08/c5a9de5aa6a146f3babf3ea0ad757f64.zip",
                "vabInfo": {
                    "data": {
                        "extra_params": "metadata_hash:d9baddba6cd82e13ae4e45a6a2020b56e00549a4ae98c1f3ebcfa4b17d34e842",
                        "header": [
                            "FILE_HASH=bMqM4X4EuPIxn0+jKCpa3bMcb7ilw3y40sv1fXKTwPs=",
                            "FILE_SIZE=7320946569",
                            "METADATA_HASH=7OIJAywApJGMdS23FLdxAjJSidPGSznQs1VeqIbesF8=",
                            "METADATA_SIZE=271505",
                            "oplus_update_engine_verify_disable=0",
                            "security_patch=2023-12-05",
                            "oplus_separate_soft=21641",
                            "security_patch_vendor=2023-12-05",
                            "ota_target_version=RMX3560_11.C.27_1270_202312042122",
                            "oplus_rom_version=V13.1.0"
                        ],
                        "otaStreamingProperty": "payload_metadata.bin:2025:271772,payload.bin:2025:7320946569,payload_properties.txt:7320948652:357,metadata:69:992,metadata.pb:1129:829",
                        "vab_package_hash": "3797b83e14892813c93b860251e18e2a"
                    }
                }
            },
            "componentVersion": "RMX3560_11.C.27_1270_202312042122.97.27ae529a"
        }
    ],
    "decentralize": {
        "offset": 18198,
        "round": 28800,
        "strategyVersion": "8"
    },
    "description": {
        "share": ".",
        "url": "https://gauss-compotacostauto-cn.allawnfs.com/remove-441ae418e0930b26531f1e13ac7d638b/component-ota/23/12/28/c887c510f50540a1b7ddf540e0a58237.html?logoType=1&fontType=opposans"
    },
    "googlePatchInfo": "0",
    "id": "658cd793124eec011ace21d5",
    "isNvDescription": true,
    "isRecruit": false,
    "isSecret": false,
    "nightUpdateLimit": "00-99",
    "noticeType": 0,
    "nvId16": "NV97",
    "osVersion": "ColorOS 13.1",
    "otaVersion": "RMX3560_11.C.27_1270_202312042122",
    "paramFlag": 1,
    "parent": "ota",
    "realAndroidVersion": "Android 13",
    "realOsVersion": "ColorOS 13.1.0",
    "reminderType": 1,
    "reminderValue": {
        "download": {
            "notice": [
                1,
                3,
                5,
                7,
                7,
                7
            ],
            "pop": [
                1,
                1,
                1
            ],
            "version": "1702974023000"
        },
        "upgrade": {
            "notice": [
                3,
                5,
                7,
                7,
                7
            ],
            "pop": [
                3,
                3,
                3,
                3,
                3
            ],
            "version": "1702974023000"
        }
    },
    "rid": "8d6c28ff-6a20-46cb-aded-324557ecb7d2",
    "securityPatch": "2023-12-05",
    "securityPatchVendor": "2023-12-05",
    "silenceUpdate": 0,
    "status": "published",
    "versionCode": 1270,
    "versionName": "RMX3560_13.1.0.400(CN01)",
    "versionTypeH5": "正式版"
}
```

</details>
Thus, the script provides information on the already installed version(RMX3560_13.1.0.400). I use the same network as the phone, I even tried adding imei. The result does not change. On the phone in the update program I see this:

[image]

mickeychen0719, 2024-01-29:
change 11.C to 11.F
realme-ota -r 1 RMX3560 RMX3560_11.F.00_0000_000000000000 4 10010111

Jipok, 2024-04-05:
@mickeychen0719 thanks

closed 2024-04-05 by Jipok