R0rt1z2/realme-ota #76: requst failed

zeng-github01, 2025-05-17:
`
 realme-ota RMX3700 RMX3700_11.F.19_2190_202503142009 6 -r 1
[2025-05-17 13:14:27.543125] I: Load payload for RMX3700 (RealmeUI V6)
[2025-05-17 13:14:27.556956] I: Wait for the endpoint to reply
Press any key to continue . . . [2025-05-17 13:14:28.624751] I: Load payload for RMX3700 (RealmeUI V6)
[2025-05-17 13:14:28.638197] I: Wait for the endpoint to reply
[2025-05-17 13:14:28.745440] F: 'errMsg'
`

mickeychen0719, 2025-05-17:
目前無法用realme-ota 查詢CN版本
只能查詢全球版
CN 版需改用updater工具查詢才可查詢

rapperskull, 2025-05-26:
Fixed by 119da37fd5e02f57c3115d7540c65855426a4798

closed 2025-05-26 by rapperskull

rapperskull, 2025-05-26:
The commit reverted #71, but I was not able to test for regression, since no example was provided.
Given that the China region was not working I don't think it matters.

bugme2, 2025-06-28:
> 目前無法用realme-ota 查詢CN版本 只能查詢全球版 CN 版需改用updater工具查詢才可查詢

updater工具是哪个，有地址吗

mickeychen0719, 2025-06-28:
realme-ota fixed, just reinstall

pip3 uninstall realme-ota

pip3 install --upgrade git+https://github.com/R0rt1z2/realme-ota

Another tool Updater
https://github.com/houvven/OplusUpdater-android